coldnew/coldnew.github.io #19: Android Things 學習筆記: Raspberry Pi 3 安裝 OS | coldnew's blog

coldnew, 2018-06-05:
https://coldnew.github.io/8e801cc4/ 

 在 Android Things 學習筆記: 介紹 一文我們介紹了 Android Things ，在我們正式開始開發我們的程式之前，我們先將手邊的 Raspberry Pi 3 Model B 裝上可以用 Android Things 的映像檔案吧。

dreamtime1122, 2018-07-24:
請問raspbeery 3  的板子也能夠實作嗎

coldnew, 2018-07-24:
這篇文章是在 Android Things 還在 preview 版本的時候寫的，當時 Google 提供的是 image 直接下載。

今年 Google 正式釋出了 Android Things 1.0, 但是下載方式變了， Google 不提供任何 Android Things 的程式碼 (像 Android Open Source Project, AOSP) 那樣，想要使用 Android Things 的話必須購買 Google 認證的核心板(System-on-Modules, SoMs)才行。

支援的硬體請見:  https://developer.android.com/things/hardware/

回到原問題，Raspberry Pi 3 是可以用 Android Things 的，你必須到 [Android Things Console](https://partner.android.com/things/console/) 建立你的 Android Things image 才行。

具體方式我再找個時間來寫 :P

a321852123, 2019-02-16:
請問我用 android things 下載下來的 映像檔都是損壞的 這是正常的嘛!?
我的系統是win10 用Win32DiskImager 燒入進SD後 老是說無法識別叫我格式化

coldnew, 2019-02-19:
@a321852123

我沒在用 windows 耶:P   你要不要把你燒完的 SD 卡放進 Raspberry Pi 3 上進行開機試試看?

Windows 應該是會去偵測卡片的 MBR 之類的它讀不讀的懂，但是非微軟的格式他有可能會讀不懂呦～

closed 2025-11-03 by coldnew