coldnew/coldnew.github.io #113: UltraZed-EG PCIe Carrier Card 開發紀錄: 使用 PetaLinux 建立系統 | coldnew's blog

coldnew, 2018-12-08:
https://coldnew.github.io/b394a9ce/ 

 在了解了如何透過 Vivado 搭配 Xilinx SDK 來控制 Cortex-A53 和 Cortex-R5 後，是時候把 Linu 裝上來啦。    針對不同嵌入式環境的需求，Xlinux 針對他們自己的平台，提供了基於 Yocto Project 製作的發行板製作工具，並命名為 PetaLinux 。這篇文章將介紹如何使用基本的 PetaLinux 功能，並建立一個可開機的映像檔 (I

adam1025, 2019-12-27:
請問 images/linux/system.bit 這個檔案是只要包含PS端的設定就好了嗎? 如果系統架好了之後產生其他的.bit檔想要執行，像是簡單的跑馬燈，要如何執行呢? 難不成要重新再做成一個BOOT.bin嗎?

coldnew, 2019-12-29:
`system.bit` 裡面是和 FPGA 相關的描述檔，因此如果你控制的 GPIO 都在 `Processing System (PS` 區域的話，是不需要一直改變他的。

但如果今天用到的 GPIO 位於 `Programming Logic` 區，就會需要產生新的 `system.bit`, 因此端看怎樣設計

adam1025, 2020-01-08:
謝謝回應，
我測試SD卡開機後如果只有新增PL部分的code其他都不變，重新產生.bit檔後只要先把他移到開發版上然後用cat xxx.bit > /dev/xdevcfg 這指令就可以順利更換.bit檔案，不用再重新build成開機檔了。
順便問一下，請問大大知道如何在petalinux產生的系統內執行c code嗎? 爬了很多文章都還是無法成功 :(

coldnew, 2020-01-08:
執行 c code ? 你是要 petalinux 幫你把 c 語言專案編譯成 binary 加入到 rootfs 還是需要將 gcc 加入到 rootfs ? 或是想要 petalinux 製作一個 sdk 讓你可以把 C 專案變成 binary ?

adam1025, 2020-01-08:
要怎麼將gcc加入rootfs呢? 我試著照https://forum.digilentinc.com/topic/12628-gcc-for-petalinux/ 所描述在prj檔案裡面輸入petalinux-config -c rootfs想要開啟設定，不過不知道為什麼我的Filesystems Package --> 裡面沒有misc這個選項..

coldnew, 2020-01-08:
可能要看你 petalinux 版本，以我用的  `2019.1` 來說，會在

```
petalinux-config -c rootfs

Petalinux Package Groups  --->
    packagegroup-petalinux-self-hosted   --->
         [*] packagegroup-petalinux-self-hosted
```

可以看到 [packagegroup-petalinux-self-hosted.bb](https://github.com/Xilinx/meta-petalinux/blob/master/recipes-core/packagegroups/packagegroup-petalinux-self-hosted.bb) 會用到  [packagegroup-self-hosted.bb](http://cgit.openembedded.org/openembedded-core/tree/meta/recipes-core/packagegroups/packagegroup-self-hosted.bb?h=master), 而 [packagegroup-self-hosted.bb](http://cgit.openembedded.org/openembedded-core/tree/meta/recipes-core/packagegroups/packagegroup-self-hosted.bb?h=master) 裡面就包含了需要的 gcc (和一堆可能你不需要的套件), 所以照我這樣打開就可以讓 rootfs 有內建 gcc 的功能

closed 2025-11-03 by coldnew